tttapa/MIDI_controller #112: [Feature request] Can you add support for piezo inputs?

marekhalmo, 2021-03-14:
Hello,
i would like to see added support for piezo inputs (piezo connected over analog in)

This is just to have a simple quiet pad that i can press. No need for velocity - i don't want to add a drum like pad

Is that possible?

I have seen that you previously closed similar issues with no explanation, so if you decline please tell me why.

Thank you!

PS. Love this lib..

tttapa, 2021-03-22:
Are you referring to https://github.com/tttapa/MIDI_controller/issues/71? That issue was closed by the person who opened the issue, saying it was no longer an issue, I didn't close it.

As mentioned in the README, this library is no longer actively maintained, all of the development happens in [tttapa/Control-Surface](https://github.com/tttapa/Control-Surface).  
I won't have time to implement this myself in the following months, but there is some discussion about piezo triggers here: https://github.com/tttapa/Control-Surface/issues/173

Wallirec, 2024-10-28:
Pieter, your example works well, but in my Pro micro 9 analog inputs and 4 and 5 do not exist. I have to initialize 11 inputs to use A10, please tell me a more correct way.

And one more question - how to assign channel 10 in the sketch?