tttapa/MIDI_controller #113: Octave change for multiplexer buttons

hajozo, 2021-06-05:
Hey,

Awesome library! Thanks.

I have built my first midi keyboard with 2 octaves (piano layout, 24 keys). I am using two multiplexers to get all the inputs for my arduino pro micro.

My first code was constructed from your  multiplexer example. I made the multiplexer pinscorrespond the midi values for piano keys. This worked perfect, but what I would love to do is have a 3- way switch to change between octaves (add or substract 12 from each key value that is given for the multiplexers buttons) I tried to make "if" and "else" functions to the code...but because the loop function only has the midi_controller.refresh command, I dont understand how to make the uint8_t variables to change keys.

I this kind of a octave change possible in the main code or does this require sone tweaks on the library?

Wish to hear from you all though this library is quite old 😁.

From Finland (sorry my bad english)

tttapa, 2021-06-07:
As mentioned in the readme, I no longer offer support for the MIDI_Controller library.

Please see <https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d2d/Transposer_8ino-example.html>.

hajozo, 2021-06-07:
This is exactly what I tried to do! Thank you very mutch!!!!!

Harri from Finland

tttapa, 2021-06-07:
For future reference: https://forum.arduino.cc/t/midi-controller-octave-changer/871731/13

hajozo, 2021-06-07:
I am hajaxo and made that thread!

Big thanks again...awesome library!